openbci-archive/OpenBCI_Python #12: How to run /scripts examples without moving the file into the home directory?

dgoodwin208, 2015-02-28:
Hi - love the code so far.

How do we run the examples in the scripts directory? 
running "python scripts/test.py" gives this:

Traceback (most recent call last):
  File "scripts/test.py", line 1, in <module>
    import open_bci_v3 as bci
ImportError: No module named open_bci_v3

This is a simple python references issue, as moving the sample scripts into the same diretory as open_bci_v3.py removes the issue.

Thanks!

jfrey-xx, 2015-03-01:
Hello,

Yes, this is mentioned in the README, you do need to copy `open_bci_v3.py` where you want to use it. "Scripts" are just example of code that you can produce without going through `user.py`, they come from the "pre-plugins era" and have been put aside to cleanup the main folder. But since `open_bci_v3.py` is not a proper library or a package, it cannot be included anywhere. We have to fix that.

dgoodwin208, 2015-03-04:
cool, thanks for the response.  
I really like the open_bci_v3.py file, am currently extending it into a server based module in a separate project. Will post promptly, would welcome your thoughts on it

commit 94d2b66d (2015-03-09): Scripts: use sys.path.append() to point to open_bci_v3.py, resolves #12
closed 2015-03-09 by jfrey-xx