openbci-archive/OpenBCI_Python #14: compile a requirements.txt file?

nat-n, 2015-03-21:
I was just trying to get started but kept running into python dependencies I didn't recognise. A requirements.txt file to be installed via `pip -r requirements.txt` would be very helpful.

jfrey-xx, 2015-03-21:
Yes, it would be a good idea.

Do you remember which dependencies you had trouble with? Numpy and yapsy are required to launch user.py, the latter being the only "exotic" lib to install as far as I recall.

nat-n, 2015-03-21:
I'm not sure exactly, I know serial was one of them. I went and installed the requirement.txt file from the OpenBCI HUB project (had to download pyOSC manually because pip wouldn't do it) then came back to OpenBCI_Python and then all the missing requirements were resolved.

Then I was able to run `PYTHONPATH=./. python ./scripts/udp_server.py --serial /dev/tty.usbserial-DN0093Y5`

jfrey-xx, 2015-03-21:
If pyOSC is not present, it should not prevent the main program (user.py) to start, instead the corresponding plugin won't be loaded. If not, it's a bug :\

Because it is unstable, you have to use the "--pre" switch to install pyOSC with pip (see README). However I only tried this on linux, not with another OS.

Note that the direct use of ./scripts/udp_server.py is deprecated (also written in the README... obviously we have to improve its readability :D).

rceballos98, 2015-06-25:
jnautly just added a reqs file!

closed 2015-06-25 by rceballos98