openbci-archive/OpenBCI_Python #17: Pip requirements not installing yaspy

helloworld, 2015-10-07:
I am receiving the error: 

    ImportError: No module named yapsy.PluginManager

I ran the command `pip install -r requirements.txt`, but the Plugin Manager does not seem to properly install. 

How do I fix this? (I am trying to run this in virtualenv on Ubuntu).

rceballos98, 2015-10-09:
Are there any error reports regarding Yapsy when you run the requirements install?
Have you tried installing yapsy on it's own (it might be the quickest solution for you).

jfrey-xx, 2015-10-13:
Ok, that's strange, I just had a try on my machine and it's not yapsy but pyOSC that caused troubles. After I changed pyOSC requirement from  "0.3.5b5294" to "0.3.5b-5294" it was good to go.

I do not have a vanilla installation, here is what has been installed by pip: numpy pylsl pyOSC pyserial requests six socketIO-client websocket-client Yapsy backports.ssl-match-hostname

(K)ubuntu 14.04, pip 1.5.4, python 2.7.6

What versions are you using?

helloworld, 2015-10-16:
It turns out that using 'sudo' makes it so yapsy cannot be found. When I took that out, it started working. 

The issue I have now is that the data isn't streaming. I know the board connected successfully, but when I input '?' like the docs suggested to get the board configuration, nothing is returned. 

Any suggestions?

rceballos98, 2015-10-16:
So the code actually searches for a '/' to capture commands to the user application. The / slash is an identifier for these types of commands. If you just write text, like '?', it'll get passed on directly to the board, as seen in the code below. 

```python
if('/' == s[0]):
	s = s[1:]
	rec = False # current command is recognized or fot
...
				
elif s:
	for c in s:
        board.ser.write(c)
        time.sleep(0.100)
```
To be clear, you have no errors or warning now when using the code? It sounds like your board might in fact not be properly connected yet.

Note: Sometimes the board might take a little longer than expected (by the program) to return the first amount of data.
Try sending the '?' command a couple of times, or just press enter to check if the board has sent anything back.

helloworld, 2015-10-19:
That makes sense, but the elif statement in the code is nested under the main `if('/' == s[0]):` conditional, so the plain text commands without a slash will never be sent to the board. 

Is this an error in the code? Or am I still incorrectly understanding this?

Thanks!

jfrey-xx, 2015-10-19:
I think you're right, it's another regression from 49b3727; I missed this one when I reviewed the code last week.

Nice catch ;)

One should be more cautious when switching between tab and spaces -- I'm looking at you @chipaudette  :p

I'm gonna fix this; in the meantime I'm closing this bug since we're not talking about yapsy anymore.

closed 2015-10-19 by jfrey-xx