openbci-archive/OpenBCI_Python #22: What is a channel?

MartinThoma, 2015-11-26:
Is a channel one electrode?

jfrey-xx, 2015-11-27:
It's not the same but basically, yes, with OpenBCI *most of the time* a channel is connected to an electrode (you could plug in fact any kind of analog sensor).

Are we using somewhere in this repository confusing terms, or... can I close this "bug"?

teonlamont, 2015-11-27:
hi @MartinThoma, fancy seeing you here :)

closed 2015-12-05 by jfrey-xx

MartinThoma, 2015-12-05:
@jfrey-xx No, it can be closed.  I'm sorry I asked it here, but I didn't know where else I should ask. And I also think this might be helpful for other people looking for the term in this context.

@teonlamont Haha, the world is small :-)

jfrey-xx, 2015-12-05:
No problem, the issue tracker is indeed helpful for those kind of messages ;)

If you have general questions about the device or EEG, don't hesitate to visit OpenBCI's forum!