openbci-archive/OpenBCI_Python #24: Trouble in the serial connection

Nikre93, 2016-05-04:
Hi,
this is the error in my terminal. I don't understand what has gone wrong, someone can help me?  

No handlers could be found for logger "yapsy"
            USER.py
user.py: open_bci_v3...

------------SETTINGS-------------
('Notch filtering:', True)
user.py: Logging Disabled.

-------INSTANTIATING BOARD-------
Connecting to V3 at port /dev/ttyUSB0
Serial established...
Traceback (most recent call last):
  File "user.py", line 98, in <module>
    log=args.log)
  File "/home/pi/Desktop/OpenBCI_Python-master/open_bci_v3.py", line 88, in __init__
    self.print_incoming_text()
  File "/home/pi/Desktop/OpenBCI_Python-master/open_bci_v3.py", line 319, in print_incoming_text
    c = self.ser.read().decode('utf-8')
  File "/usr/lib/python2.7/encodings/utf_8.py", line 16, in decode
    return codecs.utf_8_decode(input, errors, True)
UnicodeDecodeError: 'utf8' codec can't decode byte 0xff in position 0: invalid start byte

teonbrooks, 2016-05-05:
hey @Nikre93, what command did you try using with it? also what would you like to do with the data?

also, there's a forum that may be useful http://openbci.com/index.php/forum/

Nikre93, 2016-05-05:
hi... i used the command python user.py -p /dev/ttyUSB0...

I would like to light up a led with a simple EMG...

teonbrooks, 2016-05-05:
It looks like your syntax is correct for the python call. maybe @rceballos98 might have some ideas

if you know javascript, the [OpenBCI-js-sdk](https://github.com/OpenBCI/openbci-js-sdk) may be the best solution. It has everything you need to get started and an active development team in case you have questions.

i3games, 2016-06-25:
It is a python Unicode decoding error. A quick workaround is to replace all occurences 

`c = self.ser.read().decode('utf-8')`
to
`c = board.ser.read().decode('utf-8', errors='replace') # TODO`

https://docs.python.org/2.7/howto/unicode.html#

commit 8955dc42 (2017-02-25): Fixing decoding issues, closes #24
closed 2017-02-25 by jfrey-xx