openbci-archive/OpenBCI_Python #25: sample output meaning

Nikre93, 2016-06-02:
Hi everybody, i have only a question: when i run the test.py, those value that print, what rapresent?

teonbrooks, 2016-06-03:
maybe @rceballos98 might know?

teonbrooks, 2016-06-03:
my guess is values in microvolts, but I'm not 100%

Nikre93, 2016-06-03:
Those values are always negative and very high...  I think that the openBCI Gui make another processing of data.. I can wrong..

rceballos98, 2016-06-03:
Hi Nike, 
Teon is correct, they are microvolts values for the voltage difference between each electrode and the reference electrode. If they are very negative and high it might be on eof 2 reasons. 
1) that's actually the real value and what you are used to looking at from the processing GUI is one where the baseline or DC offset has been removed. This can be done by applying a high pass filter to the signal. 
2) there's an issue either with the board or the electrode placement. If the GUI seems fine though, it's probably option 1.

Nikre93, 2016-06-04:
Very Thankful...  With your input I found in github the EEGrunt that do what you say...  I'm working on..

closed 2016-06-05 by teonbrooks