openbci-archive/OpenBCI_Python #26: find_port is not defined in open_bci_v3

graeme44, 2016-07-01:
```
from open_bci_v3 import *
board = OpenBCIBoard()
```
Produces error:

```
NameError                                 Traceback (most recent call last)
<ipython-input-2-376316daaf9a> in <module>()
----> 1 board = OpenBCIBoard()

/home/graeme/devel/openbci/OpenBCI_Python/open_bci_v3.py in __init__(self, port, baud, filter_data, scaled_output, daisy, log, timeout)
     70     self.log = log # print_incoming_text needs log
     71     if not port:
---> 72       port = find_port()
     73       if not port:
     74         raise OSError('Cannot find OpenBCI port')

NameError: name 'find_port' is not defined

```

gabrielibagon, 2016-07-01:
`find_port()` is called when no port is specified at the OpenBCIBoard instantiation, probably as a way to make the program automatically detect the port in use by the OpenBCI dongle. However, it seems like the actual function `find_port()` doesn't appear in the code. 

As a workaround, just specify the port when you instantiate OpenBCIBoard: `board = OpenBCIBoard(port='COM1')`. Replace `COM1` with the name of the port you're using.

@teonbrooks and @rceballos98: Do you guys know why find_port() is there? Should we remove it and force the user to include the port parameter during instantiation?

teonbrooks, 2016-07-03:
@gabrielibagon, sorry, can't be of help, don't know this code.

gabrielibagon, 2016-07-20:
This issue is now fixed in pull request #27. Automatic serial port detection enabled.

closed 2016-07-20 by gabrielibagon