openbci-archive/OpenBCI_Python #28: streamer-LSL incompatibility with LSL's MATLABViewer vis_stream

haniawni, 2016-07-14:
Hello!

When using user.py's streamer-lsl plugin, the LSL stream gets created successfully and streams data, as tested by LSL's RecieveData.py. However, when running LSL's MATLABViewer app, while it is able to detect the stream, once it starts receiving data it immediately crashes, displaying the error:

> vis_stream error: Index exceeds matrix dimensions.
occurred in:
   vis_stream/on_timer: 135
   timercb: 30
   timercb: 13

Note that the problem is not in vis_stream, because when I send artificial data via LSL's example script SendData.py, vis_stream successfully detects it, connects to it, and displays the streaming data.

To aid debugging, here's my version numbers:

- OpenBCI 32Bit 8Channel Board V3
- Python 2.7
- OpenBCI_Python cloned July 13, 2016 (Commit: 65cb3f5141d8c1048a696bf618e9b099f9c4f64f)
- MATLAB R2015b (8.6.0.267246)
- LSL cloned July 13, 2016 (Commit: b9ff52584e0a6c9ef13de050b995a4816f82b0b9) master branch
- MacOSX 10.11.5 El Capitan

The command I'm using to run user.py:

> python user.py -p /dev/tty.usbserial-DJ00ISQW --add streamer_lsl

gabrielibagon, 2016-07-14:
This error usually comes up because of the parameters that you input in the MatlabViewer. These are the parameters that usually work for me:

[image]

Try similar parameters (especially the "Channels to display: [1:9]" part). Let me know if this resolves the issue.

haniawni, 2016-07-14:
Hello!
These parameters worked for me perfectly. It still reported this:

> The number of channels in the steam does not match the number of labeled channel records. Using numbered labels.

But proceeded to successfully stream /test3 signals (and will presumably stream real data) into the viewer in Matlab.

closed 2016-07-14 by haniawni

dwardu, 2017-06-21:
In my case the problem was resolved by selecting the “Sampling rate for display” to be an integer divisor of the signal sampling rate, e.g. if signal is sampled at 256Hz, choose display sampling rate of 128Hz or 64Hz, and not of 100Hz. Not doing this can cause [this line](https://github.com/sccn/labstreaminglayer/blob/master/Apps/MATLABViewer/vis_stream.m#L135) to fail.