openbci-archive/OpenBCI_Python #30: pyOSC doesn't work with Python 3

gabrielibagon, 2016-07-20:
I tried to install pyOSC with Python 3, and there are a lot of compatibility issues (using the version installed via pip + requirements.txt). We could rewrite parts of the module and include the file in the repository, or we could find another library that is more compatible with Python 2.7 and Python 3.

I'll be looking into a fix myself, but I wanted to shout this out there in case someone has insight.

RenaudWasTaken, 2016-10-20:
Hey !

I am actually using this with python 3 :D
I've had the same issue as you had but after using the 2to3 binary on pyOSC source
there were only a few minor fix left :D

I there a real need for python 3 if so I could probably help

RenaudWasTaken, 2016-10-22:
If anyone is interested I made a pull request here: https://github.com/ptone/pyosc/pull/7

You can clone my python3 fork of the pyOSC repo here:
https://github.com/RenaudWasTaken/pyosc/tree/python3

micuat, 2017-06-08:
It seems pyOSC has not been not maintained so you should consider switching to python-osc (they support python3 and has better API). It should be fairly simple to switch but currently I'm having issues with my sd card on raspberry pi so I won't be able to send a pull request :(

commit 661552e3 (2017-07-27): Merge pull request #43 from micuat/python3

Python3 support, fix #30
closed 2017-07-27 by jfrey-xx