openbci-archive/OpenBCI_Python #34: Cannot connect to Ganglion

matthew-groth, 2017-02-09:
Running on Mac OSX 10.12 Sierra

This is all very new to me, so I may be doing something wrong. But here is what I have tried after looking through the code:

I am using the command:
`python user.py --board 4 -p PORT`

Here are some of the things I have tried for PORT:
/dev/tty.usbserial*
/dev/tty.usbserial-*
/dev/ttyUSB0
/dev/tty.usbmodem1411

/dev/tty.Bluetooth-Incoming-Port

With /dev/tty.Bluetooth-Incoming-Port it finally worked... or maybe not. Typing "?" gave no output:

[image]

I noticed in the code of open_bci_v_ganglion.py that find_port() is not defined. I tried copying over find_port() from v3 (and removing the requirement to input a port in the beginning) but this didn't work either:

[image]

gabrielibagon, 2017-02-10:
This library doesn't support the Ganglion at the moment. If you'd like access to Ganglion data in python, I recommend using the [Node JS Ganglion library](https://github.com/OpenBCI/OpenBCI_NodeJS_Ganglion#interfacing-with-other-tools), and sending that data to python, like in this example (https://github.com/OpenBCI/OpenBCI_NodeJS/tree/master/examples/python).

matthew-groth, 2017-02-10:
Ultimately what I need is a way to program neurofeedback. My first go-to was OpenVibe, but I'd prefer to stick to MacOS if possible and currently they don't have a mac build. My second option was BCILab, which according to the docs needs Lab Streaming Layer, which again according to the docs requires this python library. Do you know if NodeJS can help me connect to any BCI software?

gabrielibagon, 2017-02-10:
It is theoretically possible to use LSL with NodeJS (although I haven't tried it yet). For example, if you're sending nodejs data to python (as linked above), you can then call LSL functions to send data to an LSL stream (here's how you call LSL in python: https://github.com/sccn/labstreaminglayer/blob/master/LSL/liblsl-Python/examples/SendData.py.

Alternatively, you can try this method, which does a similar trick using NodeJS + Python: https://github.com/OpenBCI/OpenBCI_NodeJS#interfacing-with-other-tools

Unfortunately, there's no official LSL support for javascript, so these workarounds seem to be the best way to go from NodeJS to an LSL stream.

Lastly, there is LSL support coming to the OpenBCI GUI in the next couple of weeks, which will provide LSL support for the Ganglion.

matthew-groth, 2017-02-10:
Thanks for all the ideas! I've been working on NodeJS since I saw your response, and I've run into a couple dead ends. Feel free to close this issue, and I'll make new issues on the NodeJS page. 

Also, hearing that you are adding LSL support straight to OpenBCI GUI made my night. Maybe I'll chill out a little and just use that when it is added. Can't wait!

closed 2017-02-11 by gabrielibagon