openbci-archive/OpenBCI_Python #38: /exit command doesn't work

GPrathap, 2017-04-05:
All most all the time,  `/exit` command does not work. I have to kill the program manually. Is this expected behavior?

jfrey-xx, 2017-04-05:
...no it is not. I have to say that I never use this command (...I always use C-d, shame on me :D), but for once it should not work while streaming (use /stop first). Can you describe reproductible steps? (e.g. which command did you use to launch the program, etc). It would be very handy to know in which case it works and in which it doesn't!

GPrathap, 2017-04-06:
Let me explain so as to reproduce this situation.  I created my own plugin and started streaming. All works fine. Then `/stop ` command also works as expected. But then when I try to execute /exit command it doesn't work. 

But as you said, if we just connect to board without running any plugin, then /exit command works fine.

Hope this will help you get into this issue.

jfrey-xx, 2017-05-11:
Sorry, it has been a long time. I did not investigate the issue, is this behavior occur when any plugin is enabled, or just yours?

GPrathap, 2017-05-11:
Nope, I got the same error with every plugin which I tried. To be specific, use this plugin:"csv_collect record.csv" to reproduce it. These days quite busy with ongoing works. Probably I will able to look into this issue in next month.

donkushan, 2017-06-24:
I also faced the same problem and if it's possible, I would like to work on this issue and fix this.

jfrey-xx, 2017-06-26:
Of course you are more than welcome to work on the issue and make a pull request if you manage to solve it! At this moment I don't have much time to dedicate to the various issues, any help is appreciated ;)