openbci-archive/OpenBCI_Python #39: Code don't run in windows

Danielherresan, 2017-04-19:
Please sorry my redaction, the english it's my second language.
I am trying to run this code in windows 10, after that i used Zadig tool for using my ganglion board, streaming the signal with OpenBCI GUI and this work fine. 
My goal its run in python a algoritm of machine learning using a Rasphery like processing plataform. I am familiarizate with python then its very helpfull this code. I need Know how i can run it on Windows.

A friend have linux and in this enviroment install all requirements very easy, but dont found the ganglion board in the port 0,1,2,3...30. If you can help me i will be very glad.

jfrey-xx, 2017-04-19:
Hello,

Unfortunately at the moment bluepy, the python library used to to communicate through bluetooth (BLE) does not support windows. There are alternatives to bluepy, but none seem to support all three OS (linux, mac, windows). It seems that windows 10 is particularity difficult to handle, so you may be better off directly trying to use a Raspberry.

As for your friend on linux, you must not pass a port as an option (as with the cyton), but the MAC address of the bluetooth dongle. Note also that it should be possible to auto-detect the MAC address if you don't specify any.

Good luck!

Danielherresan, 2017-04-20:
How do  you auto-detect a MAC address?

Danielherresan, 2017-04-20:
We're actually thankful  for your  previous answer, sorry for  the impoliteness,but the question just popped out of nowhere

jfrey-xx, 2017-04-20:
No worry about "impoliteness", I do the same ;)

The "auto-detect" feature is enabled by default, e.g. to print the values: `sudo python user.py --board ganglion -a print` (sudo is needed for this "auto" mode).

I agree that the help could be reworked, I forgot to mention the MAC address, I will at least add that.

Danielherresan, 2017-04-20:
Thank you for your previous answer
We tried using the command you copied but we recieve an import error saying: "No module named btle". We've already installed bluepy.
Thank you for helping us so far

jfrey-xx, 2017-04-21:
Did you install bluepy externally? You have to fetch the version that is comprised within this repository (clone with recursive option, compile, etc), see the "Ganglion Board" section in the Readme. Do no hesitate to comment if the help does not make sense or is not well phrased, we may try to improve that part thanks to your feedback ;)

Danielherresan, 2017-04-21:
Thank you for your answer,
We are very glad with your help. Indeed we clone the repository, but this don't run because the system don't recognize import btle. We solve it change import btle with import bluepy and where appears  btle we change it with bluepy.btle. Its enough with this steps we solved the problem, then thanks for your help now you can close this issue.
If in the future run in windows will be awesome :D. Congratulations for your excelent job.

PR #113 (2019-01-22): Fix cross platform issues
This implements the changes discussed in #112. Specifically removing bluepy from `requirements.txt`, adding a section in the `README` about the need to install it to use the Ganglion code, and adding a if `sys.platform == "linux"` check to import ganglion in `__init__.py` to allow all code except Ganglion to be used on all platforms.

Resolves #39, resolves #68, resolves #72, resolves #86, resolves #98, resolves #112.
closed 2019-01-22 by daniellasry