openbci-archive/OpenBCI_Python #40: Code don't run in Raspberry

Danielherresan, 2017-05-10:
Hello again, after this works in computer with linux, We try launch in a Rasphery 3, well we made the same steps but when we enter sudo python user.py --board ganglion -a print, this output "need more than 1 value to unpack". It's the same code and the same entry we dont understand.
we are very glad for your atention

Danielherresan, 2017-05-10:
This is the complete output
pi@bci:~/Downloads/OpenBCI_Python-master $ sudo python user.py --board ganglion -a print

------------user.py-------------
Board type: OpenBCI Ganglion
Will try do auto-detect board's port. Set it manually with '--port' if it goes wrong.
ERROR:yapsy:Unable to import plugin: /home/pi/Downloads/OpenBCI_Python-master/plugins/streamer_lsl
Traceback (most recent call last):
  File "/usr/local/lib/python2.7/dist-packages/yapsy/PluginManager.py", line 487, in loadPlugins
    candidate_module = imp.load_module(plugin_module_name,plugin_file,candidate_filepath+".py",("py","r",imp.PY_SOURCE))
  File "/home/pi/Downloads/OpenBCI_Python-master/plugins/streamer_lsl.py", line 6, in <module>
    from pylsl import StreamInfo, StreamOutlet
  File "/usr/local/lib/python2.7/dist-packages/pylsl/__init__.py", line 2, in <module>
    from .pylsl import IRREGULAR_RATE, DEDUCED_TIMESTAMP, FOREVER, cf_float32,\
  File "/usr/local/lib/python2.7/dist-packages/pylsl/pylsl.py", line 1147, in <module>
    lib = CDLL(libpath)
  File "/usr/lib/python2.7/ctypes/__init__.py", line 365, in __init__
    self._handle = _dlopen(self._name, mode)
OSError: /usr/local/lib/python2.7/dist-packages/pylsl/liblsl32.so: cannot open shared object file: No such file or directory

------------SETTINGS-------------
Notch filtering:True
user.py: Logging Disabled.

-------INSTANTIATING BOARD-------
Looking for Ganglion board
Try to detect Ganglion MAC address. NB: Turn on bluetooth and run as root for this to work! Might not work with every BLE dongles.
Scanning for 5 seconds nearby devices...
Traceback (most recent call last):
  File "user.py", line 120, in <module>
    aux=args.aux)
  File "/home/pi/Downloads/OpenBCI_Python-master/open_bci_ganglion.py", line 80, in __init__
    port = self.find_port()   
  File "/home/pi/Downloads/OpenBCI_Python-master/open_bci_ganglion.py", line 181, in find_port
    devices = scanner.scan(scan_time)
  File "/home/pi/Downloads/OpenBCI_Python-master/bluepy/btle.py", line 668, in scan
    self.start()
  File "/home/pi/Downloads/OpenBCI_Python-master/bluepy/btle.py", line 606, in start
    self._mgmtCmd("le on")
  File "/home/pi/Downloads/OpenBCI_Python-master/bluepy/btle.py", line 272, in _mgmtCmd
    rsp = self._waitResp('mgmt')
  File "/home/pi/Downloads/OpenBCI_Python-master/bluepy/btle.py", line 317, in _waitResp
    resp = BluepyHelper.parseResp(rv)
  File "/home/pi/Downloads/OpenBCI_Python-master/bluepy/btle.py", line 282, in parseResp
    (tag, tval) = item.split('=')
ValueError: need more than 1 value to unpack

Danielherresan, 2017-05-10:
Using Ubuntu mate, the issue is "Exec format error"

jfrey-xx, 2017-05-11:
Hello,

There is one error regarding LSL in the output, a separate issue due to the fact that there is no ARM library shipped with it with PIP, and it should not impact the rest of the program.

As for the main error, I'd say it is due to the implementation of the BLE stack on raspberry. Have you try to specify the MAC adresse of the board instead of relying on auto-detection?

Danielherresan, 2017-05-11:
hello, thanks for your answer 
Indeed we try with this command "sudo python --board ganglion -a print -port f6..." but output the same error. then we tried pairing the ganglion with the raspberry, ussing commands in the terminal in order to confirm the Mac addres and the bluetooth configurations are fine. Well we paring using gatttool, and this works. But we hope run your code in our raspberry, its awesome code and very practical. thanks again

Danielherresan, 2017-05-11:
one more question. it's better try in raspbian or in ubuntu mate

Danielherresan, 2017-05-22:
Hello, please can you help us?

jfrey-xx, 2017-05-22:
Hello,

I cannot test ganglion+Pi at the moment, so it is quite hard to debug. The output should not be the same when you specify a port. Maybe the problem comes from bluepy. Have you try to connect to the board in a separate script with python?

As for raspbian vs ubuntu mate... I don't know, I personally use the former with Pi, but I don't know if there will be a different behavior between the two.

Good luck!

Danielherresan, 2017-06-01:
Thanks for your answer
We've been working over this issue during the past few weeks in both Raspbian and Ubuntu mate. We've finally decided yo keep working in mate and now we've got this error:
INSTANTIATING BOARD-------
Looking for Ganglion board
('--->', 'F6:09:69:25:19:37')
Init BLE connection with MAC: F6:09:69:25:19:37
NB: if it fails, try with root privileges.
Traceback (most recent call last):
  File "user.py", line 121, in <module>
    aux=args.aux)
  File "/home/bci/Escritorio/PetSnake/OpenBCI_Python-master/open_bci_ganglion.py", line 85, in __init__
    self.connect()
  File "/home/bci/Escritorio/PetSnake/OpenBCI_Python-master/open_bci_ganglion.py", line 112, in connect
    self.gang = Peripheral(self.port, 'random') # ADDR_TYPE_RANDOM
  File "/home/bci/Escritorio/PetSnake/OpenBCI_Python-master/bluepy/btle.py", line 353, in __init__
    self.connect(deviceAddr, addrType, iface)
  File "/home/bci/Escritorio/PetSnake/OpenBCI_Python-master/bluepy/btle.py", line 388, in connect
    self._startHelper()
  File "/home/bci/Escritorio/PetSnake/OpenBCI_Python-master/bluepy/btle.py", line 246, in _startHelper
    universal_newlines=True)
  File "/usr/lib/python2.7/subprocess.py", line 711, in __init__
    errread, errwrite)
  File "/usr/lib/python2.7/subprocess.py", line 1343, in _execute_child
    raise child_exception
OSError: [Errno 8] Exec format error

micuat, 2017-06-06:
@Danielherresan @jfrey-xx I am working on the same setup (Ganglion + Pi 3) and the script worked like a charm. The only thing is that the script has to be run with `sudo` privilege; otherwise it won't autodetect the MAC address. Or, if you already know the MAC address, simply specify it with `-p`

jfrey-xx, 2017-06-06:
Ah, nice to see that it works for some!

Maybe it is also a matter of distribution? Which one are you running, Raspbian, Ubuntu Mate?

Thanks for the feedback @micuat, I was getting slightly desperate :D

micuat, 2017-06-06:
@jfrey-xx here you go:
```
$ cat /etc/os-release 
PRETTY_NAME="Raspbian GNU/Linux 8 (jessie)"
...
```

As a quick test I could stream data via OSC (not electrodes attached, though)

Danielherresan, 2017-06-06:
Hello, thanks for your answers. 
We applied all the steps that "micuat" described since the very first attempt. Maybe the difference is the libraries installed by micuat. If @micuat could describe all of the steps he followed since the installation of Raspbian we'd be very thankfull.

micuat, 2017-06-07:
@Danielherresan I cannot describe all the steps since I set up the pi a few month ago! For OpenBCI_Python, I did

```
sudo apt install python-dev
git clone htps://github.com/OpenBCI/OpenBCI_Python.git
cd OpenBCI_Python
git submodule update --init --recursive
cd bluepy/bluepy
make
cd -
sudo pip install -r requirements.txt
```

but I had to edit `requirements.txt` as described in issue #41.

The versions are
* Raspbian GNU/Linux 8 (jessie)
* bluez 5.23-2+rpi2
* bluez-firmware 1.2-3+rpi1
* python2.7 2.7.9-2+deb8u1

Danielherresan, 2017-06-08:
We followed your steps and it actually worked like a charm!!! Thank you very very much!!

jfrey-xx, 2017-06-08:
Good news! I guess that it is the installation and compilation o the local copy bluepy that gave you troubles. Thanks again @micuat for the help. Closing the issue, now :)

closed 2017-06-08 by jfrey-xx

jose08m2, 2017-08-31:
Hello everyone. I'm working trying to connect my openBCI with my raspberry Pi 3 B using openbci_python, i can connect the target and make some tests, but when i write star to stream data says that there's not function loaded. I have all the requirements needed for this project. Hope someone can help me, thanks 

pi@raspberrypi:~/OpenBCI_Python $ python user.py 
------------user.py-------------
WARNING: no plugin selected, you will only be able to communicate with the board. You should select at least one plugin with '--add [plugin_name]'. Use '--list' to show available plugins or '--info [plugin_name]' to get more information.
Board type: OpenBCI Cyton (v3 API)
Will try do auto-detect board's port. Set it manually with '--port' if it goes wrong.
ERROR:yapsy:Unable to import plugin: /home/pi/OpenBCI_Python/plugins/streamer_lsl
Traceback (most recent call last):
  File "/home/pi/.local/lib/python2.7/site-packages/yapsy/PluginManager.py", line 487, in loadPlugins
    candidate_module = imp.load_module(plugin_module_name,plugin_file,candidate_filepath+".py",("py","r",imp.PY_SOURCE))
  File "/home/pi/OpenBCI_Python/plugins/streamer_lsl.py", line 6, in <module>
    from pylsl import StreamInfo, StreamOutlet
  File "/home/pi/.local/lib/python2.7/site-packages/pylsl/__init__.py", line 2, in <module>
    from .pylsl import IRREGULAR_RATE, DEDUCED_TIMESTAMP, FOREVER, cf_float32,\
  File "/home/pi/.local/lib/python2.7/site-packages/pylsl/pylsl.py", line 1147, in <module>
    lib = CDLL(libpath)
  File "/usr/lib/python2.7/ctypes/__init__.py", line 362, in _init_
    self._handle = _dlopen(self._name, mode)
OSError: /home/pi/.local/lib/python2.7/site-packages/pylsl/liblsl32.so: cannot open shared object file: No such file or directory

------------SETTINGS-------------
Notch filtering:True
user.py: Logging Disabled.

-------INSTANTIATING BOARD-------
Connecting to V3 at port /dev/ttyUSB0
Serial established...
OpenBCI V3 8-16 channel
On Board ADS1299 Device ID: 0x3E
LIS3DH Device ID: 0x33
Firmware: v2.0.0
$$$
No daisy:
(8, 'EEG channels and', 3, 'AUX channels at', 250.0, 'Hz.')

------------PLUGINS--------------
Found plugins:
[ sample_rate ]
[ csv_collect ]
[ udp_server ]
[ print ]
[ streamer_osc ]
[ streamer_tcp ]
[ noise_test ]

--------------INFO---------------
User serial interface enabled...
View command map at http://docs.openbci.com.
Type /start to run (/startimp for impedance 
checking, if supported) -- and /stop
before issuing new commands afterwards.
Type /exit to exit. 
Board outputs are automatically printed as: 
%  <tab>  message
$$$ signals end of message

-------------BEGIN---------------

--> /test2
Warning: Connecting pins to low frequency 1x amp signal
%	OpenBCI V3 8-16 channel
%	On Board ADS1299 Device ID: 0x3E
%	LIS3DH Device ID: 0x33
%	Firmware: v2.0.0
%	$$$updating channel settings to default
$$$
--> /test3  
Warning: Connecting pins to high frequency 1x amp signal

--> /start
No function loaded

--> /stop
Stopping streaming...
Wait for buffer to flush...
--> /start
No function loaded

--> /stop
Stopping streaming...
Wait for buffer to flush...
--> /exit
Closing Serial...
Deactivating Plugins...
User.py exiting...

jfrey-xx, 2017-08-31:
Hi, you need to add "plugins", see help and "--add" option. By default the program will not do anything when data is acquired, not even print it. Good luck!

jose08m2, 2017-08-31:
Thank you so much, it works really nice