openbci-archive/OpenBCI_Python #42: tuning for dropped packets

micuat, 2017-06-07:
I had "dropped packets" even if I followed the instruction in README. Alternatively I did
```
sudo bash -c 'echo 8 > /sys/kernel/debug/bluetooth/hci0/conn_min_interval'
sudo bash -c 'echo 8 > /sys/kernel/debug/bluetooth/hci0/conn_max_interval'
```
and now I have very few packet lost (once in < 100 sec)

jfrey-xx, 2017-06-08:
Hum... do you think the values described in the README (9 and 10 respectively) are not "aggressive" enough? I don't know down to which point we can go... If anyone has some thought about that, we should investigate furthermore.

micuat, 2017-06-08:
I tried 9/10 at first but didn't change the situation (constantly dropping packets) so I did a google search and came across with this random project
https://gfiber.googlesource.com/buildroot/+/d5bebbc1a8c2ddb46ff609f85606cd36cf688940/fs/skeleton/etc/init.d/S31bluez.platform_gfibertv#43

So I tried 8/8 and then the issue is gone. Maybe it depends on architecture/distribution/etc? It would be nice to hear from other people about what parameters they are using...

jdpigeon, 2017-06-26:
Tried this on Ubuntu 16.0.4 x64 to no avail. With both the 9/10 and the 8/8 conn intervals I'm still getting dropped packets about twice every ten seconds