openbci-archive/OpenBCI_Python #45: Impedance testing

johnkmaxi, 2017-08-16:
@aj-ptw

I'm working on getting impedance testing implemented on the OpenBCI using Python. The fact that it was not implemented in the original OpenBCI_Python code makes me wonder if this is a more difficult task than it seems.

Does anyone have any insight on this?

Based on what I can find in the OpenBCI Processing GUI, the steps should be to
1) activate lead off negative pin for each channel of interest
2) grab the channel data coming from the board and use it for impedance calculations
3) deactivate the lead off negative pin for each channel of interest

Am I missing anything? Is anyone aware of anyone else who has got this up and running in Python?

jfrey-xx, 2017-08-18:
Hello,

I don't recall if I tried impedance testing with Python and Cyton, but if you send the proper commands to the board before typing "/start" it should give you these value, as you describe (see http://docs.openbci.com/OpenBCI%20Software/04-OpenBCI_Cyton_SDK#openbci-cyton-sdk-command-set-leadoff-impedance-commands). Once you want to switch back to regular readings, you can either type "/stop", change channels settings and type "/start", or just relaunch the whole script"

Tell us whether it works or not...

The process might be easier by adding a proper command to the main script (e.g. a "/impedance" command), hence I will tag this issue with "enhancement" if someone is willing to help.

johnkmaxi, 2017-08-18:
@jfrey-xx I went that route initially. I sent several variations of the LeadOff commands
```
z 1 1 0 Z 
z 1 0 0 Z
z 2 1 0 Z
z 3 0 1 Z
z 4 1 1 Z
```
I plotted the spectrograms of the data collected after sending those commands and it looks like those commands result in turning the channels off...

aj-ptw, 2017-08-18:
You really need to be sure to send the commands slowly with v1 firmware, update to v2-v3 and you can send them in one message.

johnkmaxi, 2017-08-18:
The board is V3-32.

By send slowly, do you mean one character at a time or one line at a time? I sent the commands like:
`z 1 0 0 Z`
after that executes, I would send the next one
`z 2 1 0 Z`

aj-ptw, 2017-08-18:
That's the hardware version. There are software versions. What do you see when you send 'v' copy and paste here please

aj-ptw, 2017-08-18:
Yes I mean send one character no faster then every 50ms. I patched this in v2

johnkmaxi, 2017-08-18:
Here it is:
```
--> v

--> 'v'
$$$
--> "v"
$$$
--> D
$$$060110$$$
```

johnkmaxi, 2017-08-24:
Ok, @aj-ptw, followed your instructions to send the instructions one character per line. The information sent to each channel was the same as indicated above:

`z 1 0 0 Z
z 2 1 0 Z
z 3 0 1 Z
z 4 1 1 Z`

After inputting each sequence of characters the message "Lead off set for channel x". I recorded some and plotted the spectrograms for each channel looking for the signal that comes through at 31.25 Hz when doing impedance testing. However, all the spectrograms were saturated at all frequencies.

Representative plots of the raw, unfiltered EEG signal and the spectrogram are shown below.

[image]

[image]
 Compared to a what I normally see in the spectrogram after doing impedance testing:
[image]

Am I still entering the commands wrong into the board? All the spectrograms look the same even though i did different combinations of PCHAN and NCHAN settings.

Narasimha-Gururaj-LTTS, 2018-02-08:
@johnkmaxi,
 Am trying to achieve the same on Cyton board. Did you succeed in getting the impedance measurement?
Also, in your previous comment you had mentioned about recording the impedance data and plotting spectrum. How did you do it?

johnkmaxi, 2018-02-09:
No, I never finished the implementation. For the spectrograms, I simply
recorded the data and plotted the spectrogram using matplotlib.

davidvumbaco, 2018-03-13:
@Narasimha-Gururaj-LTTS  - any update on your impedance efforts? We are still trying to implement these without much success.