openbci-archive/OpenBCI_Python #46: Library installation

hack-r, 2017-08-22:
Hi,

I have 2 small comments:

1. Unless I missed something, there's nothing in the README.md or elsewhere that I could find which actually says/shows how to install the library. It talks about dependencies and usage, but not installation.

2. When trying to install it like this in Debian:

`pip install git+https://github.com/OpenBCI/OpenBCI_Python.git`

It fails:

> Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-eIjxN1-build/

jfrey-xx, 2017-08-22:
Hello,

There are kind of libraries that you can include in other scripts (`open_bci_ganglion.py` and `open_bci_v3.py`) but I would not say that the repo is a proper library in itself, indeed it is meant to be used through `user.py`... and indeed there is no proper "install" section, since one should only clone the repository. That could be improved, though. If someone is used to packaging with pip... :)

hack-r, 2017-08-22:
Jeremey,

Thank you so much for the clarification and again, my apologies for the confusion.

Give me some time and perhaps I can help package it for pip, but as an R dev more likely I can help by porting this to R.

Cheers,
Jason

**********************************
Sent using Voice to Text so if there are errors, that's probably why

closed 2017-08-22 by hack-r