openbci-archive/OpenBCI_Python #47: Missing info / broken link in usage

hack-r, 2017-08-23:
Hi,

I thought I'd mention that in the [README](https://github.com/OpenBCI/OpenBCI_Python) when you get to the **user.py** section it assumes that you've connected to the board via Bluetooth, and references a broken (re-directed) link on how to do so (http://docs.openbci.com/tutorials/01-GettingStarted). 

My assumption is that the linked resource was replaced with the board-specific versions thereof, but (http://docs.openbci.com/Tutorials/02-Ganglion_Getting%20Started_Guide) I didn't see instructions in the board-specific getting started sections on how to connect via bluetooth and find a dev path like the one in your example.

I tried to figure it out by following the example code someone posted here: https://github.com/OpenBCI/OpenBCI_Python/issues/36 .

Using the last line of that, `sudo python user.py --board ganglion --add print`, first threw an error saying `no module called btle`. I believe this is because the `open_bci_ganglion.py` script doesn't work with the pip version of `bluepy`. So I edited `from btle...` to `from bluepy import btle`. That fixed that problem, but then it threw this:

<pre>
sudo python user.py --board ganglion --add print
------------user.py-------------
Board type: OpenBCI Ganglion
Traceback (most recent call last):
  File "user.py", line 66, in <module>
    import open_bci_ganglion as bci
  File "/home/jason/Desktop/OpenBCI_Python-master/open_bci_ganglion.py", line 415, in <module>
    class GanglionDelegate(DefaultDelegate):
NameError: name 'DefaultDelegate' is not defined
<

jfrey-xx, 2017-08-23:
Hello,

In the paragraph you mention, the broken "Getting started" link should refer to the Cyton.

For the ganglion, instead of modifying the source code, did you try to clone the repository with `--recursive" and type `make` as instructed by both the readme and #36 that you referred at first? You edited you message I think, do you confirm that you don't have any error by using the pip version of bluepy (you reported a `NameError: name 'DefaultDelegate' is not defined` before)?

hack-r, 2017-08-23:
Hi,

Yes thanks very much. I did find that it works without editing the file if bluepy is not installed from pip. I edited the Issue to remove that comment but you probably got the original version of my question by email.

Regarding the bluetooth connection - do you mean the part of the Cyton tutorial that refers to a built-in RFDuino? Since the Ganglion has Simplebee instead of RFduino, is it still possible to connect via bluetooth? Do you know if there's any example out there? It would be an awesome thing to add to the README but I know you are very busy.

Thanks again.

closed 2017-08-23 by hack-r

jfrey-xx, 2017-08-23:
I added you this example to the README. The problem is that the README was written mostly for the Cyton and "patched" for the Ganglion, it should be reworked to better split Cyton / Ganglion when the usage differs. 

Regarding my previous comment, the broken link belonged to an example that is meant for the Cyton, I took from granted that from the point of view of the computer it's "talking" to a serial pot, indeed bluetooth is only meaningful with the Ganglion (even if, *in theory* there is a BLE mode with the Cyton, that to my knowledge nobody really implemented).

commit 1af12918 (2017-08-23): half fix #47