openbci-archive/OpenBCI_Python #49: Is the character mapping off?

hack-r, 2017-08-24:
Is the character mapping off or is this just something weird in terms of how it's printing the status to the screen?

My understanding is that

! = turn on channel 1
@ = turn on channel 2
# = turn on channel 3
$ = turn on channel 4

Here's why I'm asking:

```
--> !

--> @
%	Activating channel 
%	1

$$$

--> #
%	Activating channel 
%	2

$$$

--> $
%	Activating channel 
%	3

$$$

--> %
%	Activating channel 
%	4

$$$

--> %

--> 1

--> 2
%	Deactivating channe
%	l 1

$$$

--> 3
%	Deactivating channe
%	l 2

$$$
4
--> 4
%	Deactivating channe
%	l 3

$$$

--> 5
%	Deactivating channe
%	l 4

$$$
%	Deactivating channe
%	l 4

```

jfrey-xx, 2017-08-24:
Are you using the Ganglion?

It must be the response from the board which appear one "step" later, and the bug is likely on the software side...

hack-r, 2017-08-25:
Yes, the Ganglion

closed 2026-10-16 by hack-r