openbci-archive/OpenBCI_Python #50: END_BYTE value for Cyton board Firmware: v2.0.0

gianlucagiorgolo, 2017-08-29:
I'm not sure if this is a known issue, but trying to stream data from a Cyton 8-channels board I would consistently get the following warning for all data packets

> <Unexpected END_BYTE found <193> instead of <192>

and of course no sample object.

Changing the value for `END_BYTE` from `0xC0` to `0xC1` fixed the issue.

jfrey-xx, 2017-08-29:
I did not look at the specs for firmware v2, this is probably a bug, indeed. We should auto-detect firmware (grabbing board info on start?) or at least have a swich...

aj-ptw, 2017-09-09:
@jfrey-xx this is really easily solved with a switch statement!

jfrey-xx, 2017-09-09:
I'm sure it's trivial; but I wanted to have a closer look at the other changes that might be around (+ I lent my ganglion, a bit hard to test anything at the moment).

aj-ptw, 2017-09-09:
@jfrey-xx well this is for Cyton! so maybe you can test! 

I added some code in #53 needs testing but @gianlucagiorgolo you can pull that pull request and see if it works for you!

jfrey-xx, 2017-09-09:
Damn, you're right, I jumped too quickly on a perfectly fine excuse :D

gianlucagiorgolo, 2017-09-15:
Ok I pulled the new branch, but there are a bunch of issues with undefined names. Here's the output of pyflakes:

> open_bci_v3.py:28: 'pdb' imported but unused
open_bci_v3.py:98: undefined name 'scaled_accel'
open_bci_v3.py:301: undefined name 'END_BYTE'
open_bci_v3.py:406: undefined name 'skipped_str'
open_bci_v3.py:407: undefined name 'skipped_str'
open_bci_v3.py:434: undefined name 'END_BYTE'
open_bci_v3.py:448: undefined name 'END_BYTE'

Regarding the one on line 98, I suppose it's just a typo in the parameter name in `__init__` (`scale_accel` instead of `scaled_accel`)

For line 301 I guess you could change
>self.warn("ID:<%d> <Unexpected END_BYTE found <%s> instead of <%s>"      
            %(packet_id, val, END_BYTE))

to
>self.warn("ID:<%d> <Unexpected END_BYTE found <%s> instead of <%s> or <%s>"      
            %(packet_id, val, END_BYTE_ACCEL, END_BYTE_AUX))

and the same goes for lines 434 and 448: `s/if b == END_BYTE:/if b == END_BYTE_ACCEL or b == END_BYTE_AUX/`

For the undefined `skipped_str`, I'm not exactly sure what that would be...

gianlucagiorgolo, 2017-09-15:
By the way, I forgot to mention that, after fixing the undefined names issues, everything seems to work fine!

aj-ptw, 2017-09-15:
Awesome could you edit the code to fix the error?

gianlucagiorgolo, 2017-09-15:
I created a pull request with the fixes, I guess we can close this issue now?

jfrey-xx, 2017-09-15:
Megred and issue closed, thanks! :+1:

commit 05679976 (2017-09-15): Merge pull request #55 from gianlucagiorgolo/aux-byte-support

Fixed undefined names issues close #50