openbci-archive/OpenBCI_Python #51: ADD: WiFi Shield support

aj-ptw, 2017-09-09:
Driver for openbci wifi shield!

Let's do it!

jfrey-xx, 2017-09-09:
It'd be my pleasure to look into that... should I have a wifi shield to get my hands on :stuck_out_tongue: 

Is the only difference the way to acquire raw data (i.e. through wifi instead of BLE)? Apart from that did the protocole change?

aj-ptw, 2017-09-09:
@jfrey-xx i think the first thing i will so is just build the device discovery part, add a function to connect, and send commands to the board, and stream data over in JSON with nano_volts. I don't really want to spend all the time on the raw mode hahah if you want to, you can take the code i write for connecting to it and such and write the code to have the board send raw. Faster sample rates are possible with raw for sure! but sample rates (500Hz daisy) and (800Hz) ganglion should be possible with JSON. Going for MVP here. Going to write these types of drivers in as many languages as i can :)

aj-ptw, 2017-09-09:
yea i think you need a wifi shield!

aj-ptw, 2017-09-09:
> Is the only difference the way to acquire raw data (i.e. through wifi instead of BLE)? Apart from that did the protocole change?

Yes! In raw mode, the wifi shield will send the classic 33byte with `0xA0` and `0xCn` where `n` is `0-F`. The ganglion over raw mode with wifi will send the same thing, just have four empty channels, but will have full 24bit resolution instead of 18/19 bit delta compression.

jfrey-xx, 2017-09-09:
Same raw format *and* a JSON version? Neat ;)

aj-ptw, 2017-09-09:
Yea the raw format is so you can pipeline to existing parsers. The JSON sends:
```son
{
  "chunk": [
    {
      "timestamp": ...,
      "data": [
         ....
      ]
  ]
}
```

Time is in micro seconds and data is in nano volts

aj-ptw, 2017-09-13:
@jfrey-xx hey i need some help/advice

I am trying to open a tcp server and have it run all from the init function of OpenBCIWifi() but the thing is the server requires an `asyncore.loop()` which is blocking. So how do i start the server and keep going with the init function?

Thanks!

jfrey-xx, 2017-09-13:
It has been a long time since I fought with TCP directly, quickly I'd say you could use a timeout parameter (if effective) to avoid anything blocking, or just use a dedicated (sub)class using Thread for opening socket and sending/receiving data. ...but you must have thought of this, no? I might be missing the point.

aj-ptw, 2017-09-13:
Well i got an MVP working, it's printing JSON data to the console. I need some help making it route nicely like yours does but it's a rough demo and more importantly you can check out how cool and easy it is!!!! AH i love it haha

aj-ptw, 2017-09-13:
> Thread for opening socket and sending/receiving data

Probably a good way to do this. This is so easy in javascript land haha

aj-ptw, 2018-01-12:
Picking this back up this weekend to add the high speed streaming that is now standard on 2.0.2

aj-ptw, 2018-01-13:
@jfrey-xx 

I am making this a legit package like the TODO requested :)

jfrey-xx, 2018-01-29:
...I did not have the opportunity to catch-up with your last dev; passing by to congratulate you, you did an impressive amount of work. With proper documentation, tests, packaging, the branch you are working on will deserve the mighty "1.0.0" milestone ;)