openbci-archive/OpenBCI_Python #57: user.py not working

LaraibMahboob, 2017-12-13:
/start T:3 I tried this and its giving an error that is "No function loaded".

jfrey-xx, 2017-12-13:
Hello,

The README file gives some examples of use and it explains how it works. You may have not selected a plugin -- i.e. what you want to do with EEG readings. For example if you want to simply print the raw values, you can use something like: `python user.py -p /dev/ttyUSB0 --add print`.

Hope it helps ;)

LaraibMahboob, 2017-12-14:
When I try to print eeg readings it gives me error "createfdContiguous failCorresponding SD file OBCI_8D.TXT".

jfrey-xx, 2017-12-14:
I've never seen anything like that, and I don't see why a TXT file is involved. Can you please share all the details: exact command you type, board and version of the firmware you are using, etc.?