openbci-archive/OpenBCI_Python #61: plugins initialization failed

qasimwasfi, 2018-04-04:
after running user.py script  ,when i add the plugins i get message" initialization failed ,could not find Fat16/32 partition".i have every thing alright ,Sdcard is inserted and format is fat 32.
  i am running this on Raspbian OS at Raspberry pie 3.python version is 2.7

qasimwasfi, 2018-04-14:
i was running user.py from python idle gui before,however error was resolved via sudo python user.py -p /dev/ttyUSB0 --add print command from raspbian terminal

jfrey-xx, 2018-04-14:
Hello,

Sorry for the late response, indeed, you cannot add plugins on the fly, the "add" is a flag to pass to the command line interface.

Being able to load/unload plugins after init could be a nice feature though!

closed 2018-04-14 by jfrey-xx