openbci-archive/OpenBCI_Python #63: raise OSError('Cannot find OpenBCI port')

digitalgreentea, 2018-04-14:
Hello, after turning on my Ganglion board and dongle, I run the command:

`python openbci_lsl.py --stream`

and it outputs the following:

-------INSTANTIATING BOARD-------
Traceback (most recent call last):
  File "openbci_lsl.py", line 56, in <module>
    main(sys.argv[1:])
  File "openbci_lsl.py", line 40, in main
    lsl = streamerlsl.StreamerLSL(GUI=False)
  File "C:\Program Files\MATLAB\OpenBCI_LSL\lib\streamerlsl.py", line 46, in __init__
    self.initialize_board(autodetect=True)
  File "C:\Program Files\MATLAB\OpenBCI_LSL\lib\streamerlsl.py", line 59, in initialize_board
    self.board = bci.OpenBCIBoard()
  File "C:\Program Files\MATLAB\OpenBCI_LSL\lib\open_bci_v3.py", line 77, in __init__
    port = self.find_port()
  File "C:\Program Files\MATLAB\OpenBCI_LSL\lib\open_bci_v3.py", line 574, in find_port
    raise OSError('Cannot find OpenBCI port')
OSError: Cannot find OpenBCI port

Does anyone know how to solve this error?

jfrey-xx, 2018-04-14:
Hello,

You have to specify that you are using a ganglion board with the flag `--board ganglion` and if the detection of the MAC address fails, you then should specify it `-p`. All the options are (hopefully) described in the help `python user.py -h` :)

closed 2018-04-14 by jfrey-xx

digitalgreentea, 2018-04-14:
How do you find the [port] for your device? I have a Ganglion.

digitalgreentea, 2018-04-14:
Also, how do I change the __init__ function in open_bci_v3.py to establish a serial connection in Windows?

jfrey-xx, 2018-04-15:
My bad, I overlooked that you are using windows: at the moment only linux
is supported, due to the python library used to handle BLE connections.
Sorry :\

If case you want to give Linux a try (through a virtual machine or by
installing a distribution on a live usb key for exemple), the "port" in
case of the Ganglion is not *really* a port, it is the mac address of the
device, that you should see when you can for bluetooth devices on your
machine.

Best,

Jeremy

parasmemon, 2019-11-22:
I am using OpenBCI Cyton board. when I use this code: 
from pyOpenBCI import OpenBCICyton

def print_raw(sample):
    print(sample.channels_data)
board = OpenBCICyton(port=None, daisy=False)
board.start_stream(print_raw)

I get this error:  raise OSError('Cannot find OpenBCI port.')
OSError: Cannot find OpenBCI port.