openbci-archive/OpenBCI_Python #68: btle.BTLEException: Failed to execute mgmt cmd 'le on'

3man1992, 2018-04-29:
Hey Guys,

I have macOS High Sierra as my host machine, and I'm running Ubuntu 16.04 in VMware. I've downloaded all requirements and got my vm to access my headphones, so bluetooth is working.

When I run: $sudo python user.py --board ganglion --add print

I get:

"Failed to execute mgmt cmd '%s'" % (cmd))
btle.BTLEException: Failed to execute mgmt cmd 'le on'

Any help on this would be great. I've used the GUI on the mac and streamed data, but I'm keen to get access via this python library. 

Thank you in advance.

L

jfrey-xx, 2018-04-29:
Hi,

You're headphone might be working thanks to MacOS (it recognizes them as headphones, create an audio output, and then the output of the VM is directed to them).

If you want to check that the VM has "raw" access to the BLE (and you would need to configure the VM to create some kind of pass through), try to execute as root `hcitool lescan`. It should list your devices, including the ganglion.

If it works, then indeed the problem is on the python script side. Just to check: did you ran `make` inside bluepy folder? Note that sometimes I had issues when I mixed using the scripts with sudo or as root (one would use bluepy from the subrepository, the other from an installation with pip). Try both methods as well and keep us updated!

3man1992, 2018-04-29:
Hi,

Firstly thank you for your reply, I've spent all weekend trying to get my openBCI board up and running with this repo but had no luck. 

I ran hcitool lescan from root and got `Set scan parameters failed: I/O error`

I found this link saying that Vmware doesn't support BLE in Linux guests on a mac host:

https://askubuntu.com/questions/806706/bluetooth-low-energy-on-ubuntu-virtualized-through-vmware-fusion-pro-on-mac

I'm confused to what to try next. I can't use MacOS because the BLE library is linux specific. But VMware (apparently) doesn't support BLE in Ubuntu.

Any advice would be much appreciated. I'm going to keep hacking at this.

Cheers,

L

3man1992, 2018-04-30:
Hi Jfrey,

Thanks for your advice.

I can confirm that with this set up I managed to get it to work:

Host: Mac High Sierra V10.13.4
Vm software: Vmware Fusion - Version 10.1.1
Guest: Ubuntu 16.04
Bluetooth dongle: cambridge silicon radio csr8510 a10

From this I found my board's MAC address and had to type it in the CLI as a command when running user.py - as the scan kept failing returning

`btle.BTLEException: Failed to execute mgmt cmd 'scanend`

zthurman, 2018-09-18:
Thank you both, I was going down a bit of a rabbit hole that led to similar environment issues with the following config:

- Host: Windows 10
- VM Software: VMware Workstation 14
- Guest: Ubuntu 18.04
- Bluetooth dongle: CSR dongle from OpenBCI site
- OpenBCI board: Ganglion v2 firmware

Other issues that had to be resolved in case this is helpful for future linux guest users who also use venv when running python projects:

- Bluepy-helper file not found in venv when running from PyCharm. Needed to actually go into the venv site packages and use make to generate the bluepy-helper output file from C that Bluepy needs.
`make bluepy-helper`
- Yapsy also seems to have dependencies on pycrypto and dpkt in order for that to function correctly, needed to add these to the project venv.
- After that I received the above error that 3man references. To resolve simply do what Jfrey suggests and run `hcitool lescan` (potentially as sudo) to get the Ganglion's MAC.
- Plug this into the command line for user.py and then get:
  `File "/home/boost/PycharmProjects/OpenBCI_Python/openbci/ganglion.py", line 153, in connect
    self.desc_notify = self.char_read.getDescriptors(forUUID=0x2902)[0]
AttributeError: 'Characteristic' object has no attribute 'getDescriptors'`
- As referenced in comment on line 152 of ganglion.py need to update Bluepy, in my venv updated to Bluepy 1.2.0 instead of the 1.0.5 suggested in requirements.txt
- After the above steps I was able to get the Ganglion talking to the CSR with the above host-guest setup.

Thanks again 3man and Jfrey! I was having a bit of a gnarly time getting my ganglion to talk to any of the OpenBCI stacks prior to reading this thread.

PR #113 (2019-01-22): Fix cross platform issues
This implements the changes discussed in #112. Specifically removing bluepy from `requirements.txt`, adding a section in the `README` about the need to install it to use the Ganglion code, and adding a if `sys.platform == "linux"` check to import ganglion in `__init__.py` to allow all code except Ganglion to be used on all platforms.

Resolves #39, resolves #68, resolves #72, resolves #86, resolves #98, resolves #112.
closed 2019-01-22 by daniellasry