openbci-archive/OpenBCI_Python #72: python on windows

Conijiwa, 2018-05-26:
I want to run this code on windows Os, and I find on the following website in DEPENDENCIES part, that it says that "To use a Windows OS, change the “__init“ function in open_bci_v3.py to establish a serial connection in Windows". However, I don't unnderstand how to change this function. Can anyone help me? Thanks for your help.
http://docs.openbci.com/OpenBCI%20Software/05-OpenBCI_Python

jfrey-xx, 2018-06-05:
Hello,

Did you try using the "user.py" script and changing the serial port to a more "windows" one -- i.e. tune the `python user.py -p /dev/ttyUSB0` example?

LarissaBruebach, 2018-12-17:
Hello,
we are currently trying to run this on Windows. Unfortunatly when trying to change the port with the command you suggested we get the error: "ModuleNotFoundError: No module named 'bluepy'". Bluepy is not available for windows.

Were you able to get it to run on windows? If so, what did you change?

Thanks in advance,
Lara

PR #113 (2019-01-22): Fix cross platform issues
This implements the changes discussed in #112. Specifically removing bluepy from `requirements.txt`, adding a section in the `README` about the need to install it to use the Ganglion code, and adding a if `sys.platform == "linux"` check to import ganglion in `__init__.py` to allow all code except Ganglion to be used on all platforms.

Resolves #39, resolves #68, resolves #72, resolves #86, resolves #98, resolves #112.
closed 2019-01-22 by daniellasry