openbci-archive/OpenBCI_Python #75: stream_data_wifi.py "failed to parse" errors

sarah-mcl, 2018-06-20:
Apologies in advance if this is not the right way to post such an issue.

When trying to run stream_data_wifi.py, I am able to connect to the board but the packets are garbled, as shown in the output below:
`Welcome to OpenBCI Native WiFi Shield Driver - Please contribute code!`
`Opened socket on 192.168.1.110:49624`
`Try to find WiFi shields on your local wireless network`
`Scanning for 5 seconds nearby devices...`
`Found WiFi Shield OpenBCI-4AAD with IP Address 192.168.1.104`
`Init WiFi connection with IP: 192.168.1.104`
`Connected to cyton with 8 channels`
`WiFi Shield to Python TCP Socket Established`
`WiFi Shield Instantiated`
`Incoming connection from ('192.168.1.104', 5672)`
`failed to parse: ?D??U??V?͆??
?????s???F??#??9??k???u?????s???G????????????um?u??T?s,??H?????????Ú?u?????{?sU??I?????P?̍??%?u?%????r???J??{????????f?uO?S??1???K?????????é?u?????j?sE??L?????????å?u?????Y?s;????M??????????Ë?uv?r??8?s?`
`No JSON object could be decoded`

I have an OpenBCI Cyton and Wifi Shield, both with up-to-date firmware.  Streaming works with the OpenBCI GUI, and I was able to get the js example (getStreaming.js) to work.

I'm running on a Mac so I was unable to install bluepy, but all other requirements are at or above the required versions. 

Any assistance would be greatly appreciated.

jfrey-xx, 2018-06-24:
Hello,

If the OpenBCI GUI is working then it is probably something wrong with the python script. It will be difficult to debug on my side, without a Wifi Shield. Which version of the scripts are you using, the master branch? @aj-ptw Had been working actively on a "1.0.0" branch, that isthe most up to date version of the code, and you should give it a try if not already.

sarah-mcl, 2018-06-25:
I was using the masters branch, but I tried the 1.0.0 branch and it works! I now get output that looks normal: 

`0`
`[0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]`
`0`
`[-0.007191427909306039, 0.005549111133707897, -0.0071619683101139435, -0.007198960447187477, -0.1540045474773106, -0.0487470535930459, -0.14287948076480397, -0.012493216990615962]`
`1`
`[-0.007196121775641653, 0.0055543190901659835, -0.007166550417727281, -0.007203497851311904, -0.1540087943087571, -0.04875130042449241, -0.14288372759625048, -0.012497620284273658]`

I had to modify the __init__.py file to not include the Ganglion code, because it uses bluepy, but this solves my issue! Thanks!

jfrey-xx, 2018-06-25:
Thanks for the feedback... at some point we should merge this branch to master ^^

Using try/catch we should be able to detect the need for Ganglion-related libraries, I will leave the bug report opened as a reminder.