openbci-archive/OpenBCI_Python #86: execute make -C ./bluepy clean error: [Error 2] The system cannot find the file specified

digitalgreentea, 2018-07-24:
I am installing bluepy and this error shows up every time I run pip install bluepy or run python setup.py install inside the bluepy folder.

[http://tinypic.com/r/2q09ks7/9](http://tinypic.com/r/2q09ks7/9)

Can anyone help?

shenjianaixuexi, 2018-12-09:
hello，have  you solved the problem ？ can U give me some advice

hack-r, 2018-12-09:
I'm not sure of the full context of this thread, just got this on my mobile email, but based on the screen shot it looked like a posaible Windows Bluetooth driver problem with bluepy. Remember, Windows has it's own proprietary Bluetooth drivers/language that's totally incompatible with open standards, bc some combination of they hate you and think they are better than you.

i know you can replace Windows drivers with Zadig (so?). that may cause you other Bluetooth problems tho just FYI. I would need to go check further on what I did on Windows in the past but it should work on Linux if that's an option or try the tool

PR #113 (2019-01-22): Fix cross platform issues
This implements the changes discussed in #112. Specifically removing bluepy from `requirements.txt`, adding a section in the `README` about the need to install it to use the Ganglion code, and adding a if `sys.platform == "linux"` check to import ganglion in `__init__.py` to allow all code except Ganglion to be used on all platforms.

Resolves #39, resolves #68, resolves #72, resolves #86, resolves #98, resolves #112.
closed 2019-01-22 by daniellasry