openbci-archive/OpenBCI_Python #87: Ganglion OpenBCI + Raspberry Pi 3 problem

basophobic, 2018-07-29:
Hi!
I'm trying to make my `Raspberry Pi 3` to work with the `Ganglion OpenBCI`  over Bluetooth using instructions from [here](http://openbci.com/community/raspberry-pi-for-eeg-acquisition/) by @micuat. Installation works without serious issues, I got my `Cyton `working and streaming using LSL, but still having trouble running the `user.py` script with the `Ganglion`.
I have sucessfully installed Bluetooth on my `Raspberry`, `hcitool lescan ` finds my Ganglion on the network, compiled bluepy from source (1.2.0) from [here](https://github.com/IanHarvey/bluepy), while the following python script runs without problems:
```
from bluepy import btle
dev = btle.Peripheral("HEX of Ganglion", 'random')
for svc in dev.services:
     print str(svc)
```
and outputs:
```
Service <uuid=Generic Attribute handleStart=8 handleEnd=11>
Service <uuid=Generic Access handleStart=1 handleEnd=7>
Service <uuid=fe84 handleStart=23 handleEnd=30>
Service <uuid=00001530-1212-efde-1523-785feabcd123 handleStart=31 handleEnd=65535>
Service <uuid=Device Information handleStart=12 handleEnd=22>
```
But when I issue the command:
`sudo python user.py --board ganglion --port <MAC ADDRESS GANGLION> --add print
`
the script errors with the compaint:
```
------------user.py-------------
Board type: OpenBCI Ganglion
('Port: ', 'f8:03:fc:d0:cf:25')

------------SETTINGS-------------
Notch filtering:True
user.py: Logging Disabled.

-------INSTANTIATING BOARD-------
Looking for Ganglion board
Init BLE connection with MAC: f8:03:fc:d0:cf:25
NB: if it fails, try with root privileges.
Traceback (most recent call last):
  File "user.py", line 119, in <module>
    aux=args.aux)
  File "/home/pi/OpenBCI_Python/openbci/ganglion.py", line 94, in __init__
    self.connect()
  File "/home/pi/OpenBCI_Python/openbci/ganglion.py", line 121, in connect
    self.gang = Peripheral(self.port, 'random') # ADDR_TYPE_RANDOM
NameError: global name 'Peripheral' is not defined
```
The `openbci/ganglion.py `script  , 

```
...
import sys
sys.path.insert(0,"bluepy/bluepy")
STUB_BTLE = False
try:
    from btle import Scanner, DefaultDelegate, Peripheral
except:
    DefaultDelegate = object
    STUB_BTLE = True
else:
    from bluepy.btle import Scanner, DefaultDelegate, Peripheral    
...
```
I tried to replace (comment out) the line  `sys.path.insert(0,"bluepy/bluepy")` with the path that has my compiled bluepy (`/home/pi/bluepy/build/lib.linux-armv7l-2.7/bluepy`), or even comment it (to use the system installed version?), but no luck. 
Any help?

aliraza19, 2018-11-11:
Did you find any solution to this problem?

wjcroft, 2018-11-15:
This issue is connected to issue #103 . Same "NameError: global name 'Peripheral' is not defined"

PR #105 (2018-12-03): Clean up BluePy import
This should resolve #87 and #103 by cleaning up the BluePy import. 

Previously trying to instantiate the driver with port as "TEST" would result in this error:
```
Traceback (most recent call last):
  File "/home/ian/PycharmProjects/OpenBCI_Python/openbci/test.py", line 3, in <module>
    driver = OpenBCIGanglion(port="AUTO")
  File "/home/ian/PycharmProjects/OpenBCI_Python/openbci/ganglion.py", line 94, in __init__
    self.connect()
  File "/home/ian/PycharmProjects/OpenBCI_Python/openbci/ganglion.py", line 121, in connect
    self.gang = Peripheral(self.port, 'random [... cut]
closed 2018-12-03 by andrewjaykeller