openbci-archive/OpenBCI_Python #91: plugin_interface

mesca, 2018-08-04:
When installing from pip, ``from openbci import wifi`` will give an error: ``ModuleNotFoundError: No module named 'plugin_interface'``. It is because `plugin_interface.py` is outside of the package.

Commenting the line ``from .plugins import *`` in ``openbci/__init__.py`` fixes the problem. I am not familiar with the plugin system, and maybe this quick fix has side-effects. Is there a better way?

tushar-c, 2018-08-14:
I hope somebody responds, I have the same issue

tushar-c, 2018-08-14:
If this works for you, please close the issue.

mesca, 2018-08-14:
Thanks, but your solution and mine are just workarounds, not real fixes. The point is that I don't see the reason for importing ALL the plugins and dependencies in the ``__init__.py`` file, when you just need the `wifi` module.

Anyway, you should probably submit a PR, since at least that makes it useable. Having to manually copying and editing files defeat the purpose of a pip package ;)

tushar-c, 2018-08-14:
I agree that having to manually copy and paste and edit files is definitely not something that the user should have to do. However, I wonder what exactly causes this issue. Is the `setup.py` not able to do a system-wide install of `plugin_interfaces` ? If so, then how would we fix that?

mesca, 2018-08-14:
Well, I think it's because `plugin_interface.py` is not into the `openbci` package.

tushar-c, 2018-08-14:
By `openbci` package, you mean the one install through `pip`, correct?

mesca, 2018-08-14:
Yes, but also the normal Python package in the `openbci` directory (see: https://docs.python.org/3/tutorial/modules.html#packages).

andrewjaykeller, 2018-10-17:
Sounds like we need to add the plugin_interface.py into the package that is published with pypi?

mesca, 2018-10-17:
Probably :)

andrewjaykeller, 2018-10-17:
any ideas how?

rivasd, 2019-01-18:
Since this basically breaks the whole package ( I can't even do `from openbci import cyton` ), I feel like this should be a much higher priority ?

ChristineZh0u, 2019-03-06:
Just going to confirm that this issue is still present as of today after installing with `pip install openbci-python`. Also had to install `pyserial` "manually" (even though it's in `requirements.txt` - has the build on PyPi not been updated for a while or something?)

shaielc, 2019-08-16:
this error occurs on a new install ( fresh vm ubuntu 18.10) same error trying to import anything from the obenbci pkg fails.

shaielc, 2019-08-16:
upgrading pip and reinstalling did the trick

works fine

molguin92, 2019-09-24:
This is still not working for me, and as @rivasd said higher up in this thread, it breaks the whole package.

```bash
$ pip list
Package        Version
-------------- -------
bluepy         1.3.0  
future         0.17.1 
iso8601        0.1.12 
numpy          1.17.2 
OpenBCI-Python 1.0.2  
pip            19.2.3 
pyserial       3.4    
PyYAML         5.1.2  
setuptools     40.8.0 
```

Running the following code still throws an error:
```python
import sys
from openbci import OpenBCIGanglion

def handle_data(sample):
    logging.debug(sample)

if __name__ == '__main__':
    board = OpenBCIGanglion()
    board.start_streaming(handle_data)
```

```bash
File "{...}/OpenBCIClient/venv/lib/python3.7/site-packages/openbci/plugins/csv_collect.py", line 6, in <module>
    import plugin_interface as plugintypes
ModuleNotFoundError: No module named 'plugin_interface'
```

rivasd, 2019-09-24:
@molguin92 I think that `OpenBCI_Python` is pretty much deprecated at this point, you should try out the `pyOpenBCI` repo/package instead, It works for bluetooth.