openbci-archive/OpenBCI_Python #107: Error: [print] not found or could not be loaded

aliraza19, 2018-12-09:
Hello,
We are connecting ganglion device with OpenBCI_Python but having problem related to plugins not found or could not be loaded.
Command : 
sudo python user.py --board ganglion --port E9:2A:73:ED:89:B1 --add print 

But it shows following error : 
"Error[print] not found or could not be loaded. Check name and requirements"

In our OpenBCI_Python folder, all plugin interfaces are present. Yapsy and all required packages are also installed in python,  but still i am not able to find the solution.
Kindly help us in solving the problem.

Best Regards,
Ali

wjcroft, 2018-12-14:
https://github.com/OpenBCI/OpenBCI_Python/issues/56

aliraza19, 2018-12-14:
@wjcroft  as you can see we are adding "print" but if we try to add any other plugin instead of print, we encounter this error. Do you have any idea about this problem what it can be related to.

wjcroft, 2018-12-14:
Have you tried Python 2?

aliraza19, 2018-12-14:
Python version i am trying on is 2.7

aliraza19, 2018-12-14:
There is a confusion, The solution provided in issue #56 whether is it related to our problem or not.

ikmckenz, 2018-12-14:
Actually I don't think #56 is related to this, it seems to be a separate issue with yapsy. Is this maybe because of a permissions issue? Somehow when running with sudo, python can't see the other folder? I'm not sure, as I can't seem to reproduce this bug.

aliraza19, 2018-12-16:
So what should be possible approach to solve this problem? 
Should we test it on total different system with different versions of requirements or there is any way to handle this problem. On Openbci forum there is a guy who sorted it out .

Link to the forum
http://openbci.com/forum/index.php?p=/discussion/762/advice-on-openbci-python-plugins

ikmckenz, 2018-12-17:
I was able to reproduce this issue! It's the same as #88. I'll make a branch to fix this and open a new pull request.

aliraza19, 2018-12-24:
Thanks ikmckenz for the support. As soon as i receive the hardware back i will test it and share the results here.

aliraza19, 2018-12-28:
@ikmckenz By changing the directory it worked and error is now solved.