openbci-archive/OpenBCI_Python #109: Implement Analog data parsing in Python SDK

daniellasry, 2019-01-04:
This was brought up in issue: OpenBCI/OpenBCI_WIFI#86
When switching to analog mode, no data is streaming from the Cyton. Example code:

`from openbci import wifi as bci`
`shield = bci.OpenBCIWiFi(ip_address = '192.168.1.141', log=True, high_speed=True)`
`shield.wifi_write('/2')`

`def printData(sample):`
`    print(sample.sample_number)`
`    print(sample.channel_data)`

`shield.start_streaming(printData)`
`shield.loop()`

This will print a whole bunch of zeroes.

This is because there is no analog data parsing in the Python SDK. See the empty function here: https://github.com/OpenBCI/OpenBCI_Python/blob/069bbcb9167261d687ec37206d1d2f59ace137d8/openbci/utils/parse.py#L140-L141

The implementation should be similar to the one in the NodeJS code. See [here](https://github.com/OpenBCI/OpenBCI_JavaScript_Utilities/blob/0b7e19904b6746bd0d67278d05122211edef96c8/src/utilities.js#L1366), [and here](https://github.com/OpenBCI/OpenBCI_GUI/blob/a71bd55c26a6945666d704322b77e2263f2b63fc/OpenBCI_GUI/DataLogging.pde#L321)

wjcroft, 2019-01-15:
Daniel, thanks for looking into this. Mentioning @omarclaflin .

andrewjaykeller, 2019-02-13:
The API for this should look something like

```python
from openbci import wifi as bci
shield = bci.OpenBCIWiFi(ip_address = '192.168.1.141', log=True, high_speed=True)
shield.analog_mode_on()
```

or

```python
from openbci import wifi as bci
shield = bci.OpenBCIWiFi(ip_address = '192.168.1.141', log=True, high_speed=True, analog_mode=True)
```

omarclaflin, 2019-02-13:
Sweet! Can't wait to try this later this weekend,

Thanks

daniellasry, 2019-02-14:
hey @omarclaflin, it's not implemented yet, but it will be by next week. I'll keep you posted.

omarclaflin, 2019-02-19:
Not sure if anyone else has this issue with the new code but the sample.analog_data is empty:

**Code example:**

def printData(sample):
    print(sample.sample_number)
    print(sample.channel_data)
    print(sample.analog_data)
    print(sample.digital_data)
    print(sample.aux_data)

 shield = bci.OpenBCIWiFi(ip_address = '192.168.4.1', log=True, high_speed=True, aux_mode='analog')
 shield.start_streaming(printData)    
 shield.loop()

I put in all the 'sample.xxxx's just to be comprehensive. I get:

159
[-0.18750002235174446, -0.18750002235174446, -0.18750002235174446, -0.18750002235174446, -0.18750002235174446, -0.18750002235174446, -0.18750002235174446, -0.18750002235174446, -0.18750002235174446, -0.18750002235174446, -0.18750002235174446, -0.18750002235174446, -0.18750002235174446, -0.18750002235174446, -0.18750002235174446, -0.18750002235174446]
{}
{}
{'lower': bytearray(b'\x00\xa0\x00#\x00\x00'), 'upper': bytearray(b'\x00\xa1\x00#\x00\x00')}
161
[-0.18750002235174446, -0.18750002235174446, -0.18750002235174446, -0.18750002235174446, -0.18750002235174446, -0.18750002235174446, -0.18750002235174446, -0.18750002235174446, -0.1690088935505025, -0.1711612994863152, -0.1714836786965941, -0.1739216952230567, -0.18750002235174446, -0.18750002235174446, -0.18750002235174446, -0.11265194268845828]
{}
{}
{'lower': bytearray(b'\x00\xa0\x00"\x00\x00'), 'upper': bytearray(b'\x00\xa0\x00#\x00\x00')}

daniellasry, 2019-02-19:
hey @omarclaflin , thanks for testing it out. hang tight, the code is still a work in progress!