openbci-archive/OpenBCI_Python #117: Cyton, Wifi Shield, Python (wifi.py): bad raw sample values / but fine using GUI

cmello418, 2019-02-22:
I'm using a Mac (macOS 10.14) with Python 3.6.  I have a Cyton board coupled with the WiFi Shield - 8 channels.  As of the date of this post, I have the latest firmware installed on both the Cyton and the Wifi Shield.  Everything is configured and attached to the Ultracortex "Mark IV" headset using dry electrodes.  With the latest version of OpenBCI GUI (4.0.3) everything works well.  All 8 channels stream as expected.

However, I have a problem with using the wifi.py for streaming in Python (3.6).  First, I want to point out that the square wave test works fine - I'm able to stream and display in realtime using matplotlib.  When I switch over to streaming the raw data, it seems to "work", but it does not seem to contain any real "data"; I'm getting consistent sine waves from all the channels - almost seems like test data.  Also, the values are very small float values, like .0283746734, etc.  I notice that when I jiggle my head (while wearing headset), the waves look more like you would expect, but the "normal" stream for everything is very consistent sine waves. 

I provide more details, images, and code here: https://openbci.com/index.php/forum/#/discussion/2012/cyton-wifi-shield-python-wifi-py-bad-raw-sample-values-but-fine-using-gui

cmello418, 2019-03-06:
This does not work because it isn't finished, and there are a few bugs as well.  While I don't have the time to really do this, I have no choice.  After getting roped into this, believing that it worked, I built too much code around it to redo in something else.  My PhD dissertation is now riding on this; I burned too much time thinking I was doing something wrong.  Next time, people... CLEARLY STATE that it's not finished (nor tested) in comments AT THE TOP of the code - and in documentation!

cmello418, 2019-03-09:
Changes that will soon be available in the git repository:

(1) Scale factor bug fix according to https://docs.openbci.com/Hardware/03-Cyton_Data_Format#cyton-data-format-binary-format
(2) OpenBCIWiFi now applies the sample rate passed into the constructor
(3) New option to prevent OpenBCIWiFi from automatically connecting to board upon instantiation
(4) scaled_output and micro_volts can now be set in OpenBCIWiFi constructor
(5) prevent connection exception during OpenBCIWiFi instantiation by placing small time delay between shield discovery and connecting to the board
(6) New test to demonstrate how these changes/new options work

A pull request was made; however, to get at fixes now, you can go to my fork: https://github.com/cmello418/OpenBCI_Python

closed 2019-03-09 by cmello418

wjcroft, 2019-03-09:
@cmello418, hi. Perhaps this should not be closed until integrated to master? And maybe some doc changes. Thanks so much for initiating this repair and upgrade. Sorry that the OpenBCI Python wifi code was not properly documented as "incomplete".

https://openbci.com/index.php/forum/#/discussion/2012/cyton-wifi-shield-python-wifi-py-bad-raw-sample-values-but-fine-using-gui

Mentioning @andreaortuno @produceconsumerobot

cmello418, 2019-03-09:
Thank you for your help, @wjcroft.  No apologies needed by anyone... I was a bit frazzled and perturbed at the time, and I was a bit sharp/harsh.  I know how it goes when we're all being pulled in different directions at the same time; happens to me, too.  For now, I will keep this closed, since anyone can get to my fork for a fix to this issue.  I'll make a final comment, once it's confirmed that these changes are officially merged.  Title change is a good suggestion.

wjcroft, 2019-05-03:
@andreaortuno @produceconsumerobot @daniellasry Just added on Trello checklist:

Complete pull request and code review / test from cmello fixing a number of bugs and incomplete code in the wifi.py implementation https://github.com/OpenBCI/OpenBCI_Python/pull/118