openbci-archive/OpenBCI_Python #120: Error when setting timeout on /start command

BeeCoughs, 2019-04-22:
I keep getting an error when running user.py and calling the /start command with a timeout argument (as in the tutorial)

Currently using a Cyton (no Daisy) with firmware v3.1.2

user.py plus the print plugin starts fine. Then when I try to run the start command I get an error that forces user.py to exit:

```
-------------BEGIN---------------

--> /start T:3
Traceback (most recent call last):
  File "user.py", line 232, in <module>
    lapse = int(s[string.find(s, "T:") + 2:])
AttributeError: module 'string' has no attribute 'find'
Closing Serial...
Deactivating Plugins...
Plugin PluginPrint deactivated.
User.py exiting...
```
As long as I run without the timeout argument, it seems to be fine.
Any ideas what might be going on? Happy to provide more info if needed.

daniellasry, 2019-04-26:
I'll look into this if I can reproduce the issue. What version or python do you have?

BeeCoughs, 2019-04-26:
I'm using Python version 3.7.2

daniellasry, 2019-04-27:
That's probably the issue right there. At some point we will have a python module that works with 3.x, but that's at least a few months away. The current python module is written for 2.7.

Try again with python 2.7 and see if that works.

If you must python 3, and you're feeling adventurous, you could try porting the Cyton code to python 3.

BeeCoughs, 2019-04-30:
Thanks, that worked!

Apologies, this led me to another question - in the README under "Useful Commands" there is a bit of code showing how to write a test signal (square wave) onto the SD card for 3 seconds. When I run those same lines I get "no function loaded" after the start command, I assume because there were no plugins added as arguments when running user.py. However, the example also loads no plugins. My SD card has the file that was created by sending an 'a' over serial, but it is empty (despite running '/start T:3'). Should I expect data there, or is the example perhaps from an older version of the code and the doc needs to be updated? 

Thanks again for your help.